ArnieX/swifitch #1: Possible to have the SPI pins available

osos, 2017-03-26:
I really love this work. I have been looking for this diy-HW for a long time. Thank You.

I am interested in applying i2c and/or SPI sensors, displays etc.

Would it be possible to make al the pins available through headers?

ArnieX, 2017-03-26:
Hi,

first of, thanks for the appreciation. To the i2c that's simple as ESP8266 has only SW i2c and you can set whatever GPIO pins you wish I guess default are GPIO 0 and 2 or 0 and 4, I'm not very sure now, those pins are indeed broken out on the header. About SPI I'm really short of knowledge here as I have never used this interface, in case this is too supported only in SW you could likely just set pins you wish to use within your FW.

To keep small footprint we have decided to break out only few pins for additional relay or two or some motion sensor etc. that can extend device's functionality. Breaking all pins would require a lot more space.

Alternatively if SPI pins aren't broken out or possible to set within FW. You could solder wires directly to ESP module in place of SPI pins.

Hope this helps ;), also if you make Swifich, please post pictures [here](https://github.com/ArnieX/swifitch/issues/2).

osos, 2017-03-27:
Hi,

Please see this http://www.pighixxx.com/test/portfolio-items/esp8266/?portfolioID=360

The pins 12, 13, 14 and 15 represent the hardware SPI.

i2c are normally on pin 4 and 5.

ArnieX, 2017-03-27:
Hi,

pins that you are referencing as HW SPI aren't real GPIO, these are only to directly communicate with internal memory. For example ESP8266 12S already do not have these pins as they are useless. Pin 15 is GND.

BR,

Martin

osos, 2017-03-27:
Ok. Not sure I completely understand. Just know that for my SPI enabled sensors etc, these IO-pins are to be used with the libraries available.

Would be fantastic if all available pins could be available on the PIN header on the board.

ArnieX, 2017-03-27:
That is virtually impossible due to size constrains. This device isn't to be allrounder it is meant to switch things on and off and offer limited pin out for sensors or another relays. There is currently no plan to extend the header. The best you could do is to wire pins that you need directly from ESP module by soldering wires to pins directly.

I'm sorry I can't help more here.

BR,

Martin

closed 2018-01-23 by ArnieX