ArnieX/swifitch #2: Photos of finished modules

ArnieX, 2017-03-26:
Hi everyone, please add pictures here in case you manage to make Swifitch yourself. Thank you!

LubergAlexander, 2018-02-02:
[image]
[image]
[image]

Just have finished another module - here are some pics :)

ArnieX, 2018-02-02:
Wow, nice pictures!! Thank you. Just one tip thought, wire terminals have groove so they can and should be connected together before soldering and will be nice and flush together.

LubergAlexander, 2018-02-02:
Thanks for the tip! I haven't noticed the grove first, will interconnect them before soldering next time.