ArnieX/swifitch #3: How to control multiple relays?

KrishnaPG, 2017-07-08:
Thanks for sharing this good work. It is helpful.

Wondering if you have this design extended for controlling more than one relay. And also, how to make it work with dimmers etc?

ArnieX, 2017-07-08:
Hi, this is what header is for. You can connect extra accessories to the board. If you would like it to control big relay board that should be possible if this board would be equipped with I2C. Dimmers would work too but for that I recommend dimmer that is referenced in “What’s next” in readme.

KrishnaPG, 2017-07-08:
Thanks for quick reply @ArnieX Will look into the header functionality. Thanks again for sharing this good work.

ArnieX, 2017-07-08:
You are welcome, hope you will like it if you happen to build few.

KrishnaPG, 2017-07-08:
In the folder which file is the board file? I am trying to experiment with the board to make small changes, such as using nodemcu (in place of esp), and other options such as using mounting headers instead of having to solder etc. Could not find the source files

ArnieX, 2017-07-08:
Sorry, no source file available. Using NodeMCU is overkill. Anyway NodeMCU is good for prototyping, this is actually finished thing. If you want to use NodeMCU the best would be to use other ready made modules, like relay module, etc.

KrishnaPG, 2017-07-10:
Thanks. When sending this design for PCB manufacturer (local) for a quote, they are asking for BOM. Wondering if there is BOM csv file that you can share?

ArnieX, 2017-07-10:
Well, I'm not sure. PCB manufacturers usually need gerber files. I don't know what is BOM, but I can ask my colleague who did the actual PCB design, if he knows what this is.

KrishnaPG, 2017-07-10:
BOM is short for "Bill of Materials". They would need it to assemble components on the PCB (after the creation), and also to give the estimate/quote. Essentially it tells which component goes to which location (on the PCB) and how many needed and what is the exact component brand/value etc.

ArnieX, 2017-07-10:
Hi, this information is in readme. You can therefor assemble BOM file yourself ;)

KrishnaPG, 2017-07-10:
Thanks. Wish I were that expert :)
The system (that generated the gerber etc.), must have auto-generated all these BOM automatically.  It follows certain format. Besides, I want to make sure that their placement mappings (which is not provided in the readme) is correctly captured. Request you to kindly check with your mentioned Friend - he must already be having this BOM file. It would be great help.

KrishnaPG, 2017-07-10:
Also, the placement mapping of these components is important. For example, there are multiple resistors listed in the parts. The generated BOM file keeps the mapping correct.

ArnieX, 2017-07-10:
Just one question. Are you going to make it in big amounts? If not we recommend you to solder manually.

KrishnaPG, 2017-07-10:
Not big amounts. I am more of software guy, so I am asking my friends to help me build samples. In that process, they asked me to get the BOM. It helps them to recreate the board correctly. Even if they have to solder it manually, the would need the mapping (for example, R1 is so and so resistance, C1 is so and so component etc.). Hence the request.

ArnieX, 2017-07-10:
Then again, exact mapping table is included in readme. This should be more than sufficient to do exactly what you need. I use it too when soldering.

closed 2017-07-10 by KrishnaPG