ArnieX/swifitch #4: PCB design files?

ftruzzi, 2017-08-17:
Hi!

Congrats for this project which looks very clean and well-designed. I was thinking of building something similar myself but I couldn't do it alone and it looks like you already did it!

I see your project is open hardware but I don't see any design files. Can you release them? I would love to add a second relay so that it could still fit into the wall and turn roller shutters up and down (requires two relays).

Another suggestions that comes to mind is to use a 220V - 3.3V AC-DC converter (such as [this](https://www.aliexpress.com/item/ac-dc-3-3v-supply-power-module-110v-220v-264v-to-3-3v-0-9a-small/1976378645.html)) with a 3V-switched relay and get rid of the voltage regulator, but probably 3V relays are not as popular as 5V ones.

ArnieX, 2017-08-29:
Hi,

sorry for later reply.

220V-3V3 AC-DC wasn't used as it is more expensive and not as stable as ESP might require, but you are free to test it out ;). I'm not sure how about 3V3 relays but they are less popular as most of them are sold for Arduino hackers and those require 5V.

Design files are now added to repository.

Martin

ArnieX, 2018-01-23:
Hi @ftruzzi,

have you done some changes to the board yet? Feel free to share. Thanks.

BR,

Martin