ArnieX/swifitch #5: Doubt

hkauroberoi, 2017-09-23:
Hi,

The project is simply awesome and I'm in process of ordering the PCB's. However, I'm confused when you say "_Keep in mind that each board are actually two swifitches ;)_ as there's only one relay or am I missing something. 
Further, the analog switch function - does this mean I can use a conventional wall switch to control the load along with web control? 
Thanks

ArnieX, 2017-09-23:
Hi,

thank you for kind words ;).

- 1 PCB board is actually two boards that you will break in middle, this is information for people to not order more than they really need as when you order 10pcs of PCB it will last for 20 individual modules.

- Yes you are right, 1 swifitch is 1 relay so you can control 1 thing

- Analog switch function is great if you have full N/L in under your switches at home. Usually wall switch only connect L between power source and light. N is usually not available. If you have separate N/L lines under your wall switches you are good to go with this function. You will connect the physical switch with low voltage line and one resistor to Swifitch header.

Good luck with building it and if you will remember, share your photos and experience here on GitHub, thanks.

hkauroberoi, 2017-09-23:
Thank you so much for such a prompt response. 
The analog switch function is simply great - hats off to you !! I'll be more than happy to post the images.

Thanks.

closed 2017-09-23 by hkauroberoi