ArnieX/swifitch #7: PCB design in ascii format?

dicastro, 2018-01-17:
Hi,

I am trying to create and use a swifitch, this is an idea I already had in mind, and looking for an open source/hardware solution, I have arrived to your github repos. I think your solution is a really great solution, and a ready to use solution in a real scenario (not a prototype).

I am a software developper and my knowledge about hardware and soldering skill is limited. So I am trying to import your design in this site: https://easyeda.com. In this site you can design a circuit, choose the components, design a pcb and after that you can print the pcb and buy all the components easily (I am not sure if they also send you the pcb with the components assembled, that should be great). This way it will be super easy to buy all required components for N swifitchs.

This site (https://easyeda.com) supports importing an Altium project. I am trying this option, using your published design files, but I get following error:

`Bad file format, ensure your file is Altium ASCII v5.0 format (*PCB/Schematic only*)`

1. Could you please add to the repo the design files in this format?
2. Do you know an alternative site to mentioned one, that is compatible with your design files, to do the same thing: print the pcb and buy all the components at once
3. The mentioned site also allows to share a design, it should be great that you publish your design (choosing the right components) in this site. While your are designing and choosing components you can get easily the BOM and the cost of all components to get an idea of total cost. I know that requesting this is too much, but I had to try!

Thank you very much for sharing this awesome project and making life easier to software developers with limited hardware knowledge but with great ideas.

ArnieX, 2018-01-17:
Hi,

thanks for taking time to write in. I think you won't find better documentation in either OpenSource Hardware projects. It is very straight forward to buy all the items you need. There are amounts per swifitch and links to shops where you can easily order all of it. PCB is very easy and cheap to order via seeedstudio. Soldering was done by many beginners yet and all I'm in contact with were able to finish it properly ;). So I strongly think, you should be able to.

I will have a look at provided URL but I'm pretty sure that it will come at cost. No-one will be able to provide you with PCB and complete set of parts without asking for some extra money for the time to collect it. Not even mentioning if they will do the soldering for you. Those services are often very expensive.

About the file format, I will ask my friend who did the design if he knows how to get you what you need for easyeda.com.

BTW before swifitch I had zero HW knowledge and not even great SW one. So if I could do it, so can you and anyone else ;). With this in mind I tried to do the documentation as precise and as easy to follow as possible.

I'm happy you like it. Will be back with info you requested if I can get it.

dicastro, 2018-01-17:
In EasyEDA while you are designing the circuit, you can link a component (ex: a 1 ohm resistor) to a concrete resistor offered by a concrete provider. In such a way when the circuit design is finished you have the list of all specific components offered by providers and their corresponding price.

EasyEDA has also a component store, so the sell an extensive list of components. Once the circuit is finished (with all components linked) you can make an order of all components in one click!

Ok, I'll wait for your answer concerning the PCB design in ASCII format.

ArnieX, 2018-01-17:
Well I did quick research and EasyEDA isn't offering many crucial parts. Fuses (temperature, shortcircuit), varistor, AC-DC power source, LDO regulator, ferrite bead, some capacitors etc., most other parts they have but at highet price tag than could be purchased from other sources. So I do not think, they are going to provide you all parts and if so, not at a good price.

dicastro, 2018-01-17:
Thank you very much for taking time to review EasyEDA, I really appreciate it.

I will take into account your research and I will purchase the components in other sources.

ArnieX, 2018-01-17:
Well it's up to you anyways ;), it's just my opinion, that it would actually be a lot more struggle than following documentation as others did successfuly.

dicastro, 2018-01-17:
I was trying to find a _one click_ solution and to avoid _AliExpress_ and their very long shipping times.

closed 2018-01-17 by dicastro

ArnieX, 2018-01-17:
Yeah I understand, but that is unfortunately inevitable ;), we all went this way and survived. Imagine how long it took to prototype with those shipping times. Anyway for PCB you will wait around two weeks and most of the parts from Ali will arrive in two to three weeks if you order now. Watch for Chinese New Year which is about to begin in february, then the wait could be longer. If you order now you should have all parts mid or end of february. That is still ok.

If you have additional questions or some parts links are outdated (which it shouldn't but can happen), let me know and I will assist you as much as I could. When you finish swifitches just post some pictures here in issue which is asking for pictures. It's great to see people having fun and making those.

dicastro, 2018-01-18:
Today I have purchased everything. I have bought pieces enough to build a few swifiches!

So in few weeks I hope to have received everything and to be able to assemble it successfully.

I have a 3D printer, so meanwhile I am going to print the box

ArnieX, 2018-01-18:
Awesome! Yeah 3D printer is a big helper in many projects. I have Prusa i3 MK2S. Let me know then.