ArnieX/swifitch #8: Zmenšení (Swifitch Mini)

iChUdY, 2018-02-24:
Ahoj,

v první řadě bych chtěl moc poděkovat za to, že takový projekt vůbec vznikl a byl dotažen do funkčního konce. Opravdu tleskám ve stoje a klobouk dolů. ESP8622 je prostě super chip a určitě si vaší variantu někdy zkusím sám sestavit.

U mě toto pobláznění vzniklo při koupi ESP8622 produktu Sonoff Basic, který je sám o sobě úplně o ničem, ale s vlastním firmwarem, mqtt, homebridge, HomeKit, Apple TV je to pro mě nejzajímavější produkt IoT historie. V ČR se dá koupit za 229,- (i s krabičkou 😉) z banggood za 109,- free ship👍🏻

A teď k podstatě této issue. Nejde ani tolik o to, že Sonoff vyjde levnějš a s vynaložením mnohem menšího úsilí(přidaná hodnota radosti z úspěchu sestavením vlastního ESP8622 spínače je nevyčíslitelná), ale Sonoff je prostě a jednoduše mnohem menší a většinou se vejde za běžní nástěnný vypínač(GPIO14 má samozřejmě taky).

Plánuje se varianta Swifitch Mini ?

Ještě jednou moc díky za vaší práci a přeji mnoho dalších úspěchů.

S pozdravem Standa ✋🏻

Fotky z mého flashování Sonoff Basic:
[image]
[image]
[image]
[image]
[image]
[image]

ArnieX, 2018-02-24:
Ahoj Stando,

díky za pochvalu, moc nás to těší. O Sonoffu jsme samozřejmě věděli i v době kdy jsme dělali Swifitch ale naším cílem bylo udělat hlavně bezpečný modul a superjednoduše sestavitelný i pro úplné laiky. Nějaké pojistky na sonoffu jsou také ale nevím jak moc je to celkově bezpečné. Jinak u takového zařízení jako je Sonoff rapidně klesá množství lidí, co takový modul dokáží sestavit a tak aby si neublížili, například ten zdroj 230VAC->5VDC (nebo na kolik to stepdownují) je prostě složitější postavit. O menším modulu jsme se bavili ale v současné chvíli pracujeme na něčem mnohem úžasnějším. Je možné že se ke Swifitchi časem vrátíme a uděláme upravenou verzi. Jinak jsou uživatelé Swifitche co jej mají zabudovaný pod vypínači také a funguje jim to skvěle. Rád bych tě teď potěšil informací o Swifitch Mini verzi ale zatím bohužel ;). Každopádně určitě si Swifitch postav, je to parádní pocit udělat něco svého a pak se pochlub výsledkem.

Martin Doubek

iChUdY, 2018-03-01:
Ahoj,
jsem zvědav na čem úžasným pracujete. Těším se na výsledek.

Našel jsem asi nejmenší esp switch zde:
http://tinkerman.cat/the-espurna-board-a-smart-wall-switch-with-power-monitoring/#lightbox-gallery-crvo/0/

...používají stejný zdroj i SoC a navíc ještě s HLW8012.

ArnieX, 2018-03-02:
Také pěkné ;), je toho celkem dost. Netvrdím že jsme udělali naprosto unikátní zařízení.