ArnieX/swifitch #9: How to desolder ESP8266 from NodeMCU

dicastro, 2018-02-27:
My soldering/desoldering knowledge is very limited. How could I desolder ESP from NodeMCU?

Do you have any recommended tool? Do you have some tips? I have already tried with a normal soldering tool, and it seems impossible to me.

ArnieX, 2018-02-27:
You need hot air for that ;) and do it very carefully.

LubergAlexander, 2018-02-27:
@dicastro I have tried to desolder it with a regular soldering iron and killed the chip and still wasn't able to desolder it, so a heatgun is a must :) If you have an arduino around - you can try flashing with it: https://github.com/ArnieX/swifitch-software#how-to-flash-with-arduino-uno

dicastro, 2018-02-27:
I've found this video (https://youtu.be/zwp9E37ioVw) where it is explained how to make the connection without desoldering esp from NodeMCU. I'll try this option. I do not want to buy a heatgun just to desolder one thing.

dicastro, 2018-03-21:
Finally I would like to desolder ESP from NodeMCU. I bought a heatgun, but after 10 minutes of use it is broken.

*Which brand of heatgun do you recommend me?* I would like one that lasts more than 10 minutes!

I do not know I the ones that I've seen in Amazon are good quality ones (the one that I bought was from Amazon).

I have looked in TME site and I've found these ones (there are more but I do not want to spend a lot of money):

- [SOLDER PEAK SP-1011DLR](https://www.tme.eu/gb/details/sp-1011dlr/hot-air-stations/solder-peak/)
- [ATTEN AT850D](https://www.tme.eu/gb/details/at850d/hot-air-stations/atten/)
- [THERMALTRONICS TMT-HA200](https://www.tme.eu/gb/details/tmt-ha200/hot-air-stations/thermaltronics/)
- [THERMALTRONICS TMT-HA300](https://www.tme.eu/gb/details/tmt-ha300/hot-air-stations/thermaltronics/)
- [ZD-912](https://www.tme.eu/gb/details/zd-912/hot-air-stations/)

*Do you know any of these ones?*

dicastro, 2018-03-21:
I've already bought one, so I close the issue.

Cross fingers!

ArnieX, 2018-03-21:
:D sorry I didn't reply, I was so busy today. But I wouldn't help you anyway as I do not have any heatgun except gas one, which is not really good for this job.

dicastro, 2018-03-24:
Hi again,

I still have a problem with the power of swificth.

What I've done:

- I have desoldered esp8266 form nodemcu, using my brand new soldering station
- I have connected the wires between nodemcu and swifitch as your diagram says
- I have flashed swifitch firmware
- I have powered swifitch plugging nodemcu to a USB of my laptop
- **Swifitch does not work**
    - The led gets red
    - I see the AP
    - When I try to connect from my laptop to de AP, the led starts blinking and it never connects

**However, if I power the swifitch plugging nodemcu to a mobile phone charger, it works.**

So I supose that the USB of my laptop has no current enough. **I am really interested in powering nodemcu from my laptop** to be able to see serial traces to debug the code (I am doing some changes in code).

- Do you know why with my laptop it doesn't work? (It is a brand new one, less than a year)
- Have you ever had this problem? Is there a workaround? Can I use an external power source, and connect swifitch to the laptop to see traces?

I am desperated!!

dicastro, 2018-03-25:
Thank you for your help!

I've tried with 2 laptops (both with windows) and with one raspberry pi (activating max_usb_current configuration option). And always with the same result. I am not able to connect to swifitch AP.

I've tried to create a "hello world" firmware (without any wifi connection) and everything works perfectly. I can power swificht plugging nodemcu to my laptop, and I am able to see serial traces. When I modify "hello world" firmware to connect to my home wifi, it doesn't work.

Tomorrow I am going to test swifitch at work, there I've got an iMac, and I have some collegues with MacBooks.

LubergAlexander, 2018-03-25:
I had exactly the same problem after trying to desolder the esp from
nodemcu with a soldering iron:) I have overheated and damaged the chip, it
did boot up initially, and I was able to run non wifi code, but if I tried
to connect to my AP - it crashed.

dicastro, 2018-03-25:
The ESP that I've used in the swifitch is not the one that I've desoldered from NodeMCU. So I think I do not have the same problem.

I will try to explain myself better.

Case A (firmware without Wifi connection)

```
// Case A.1
Swifitch <-> NodeMCU <-> Laptop USB

// It works but It is not stable, sometimes it reboots

---

// Case A.2
Swifitch <-> NodeMCU <-> Mobile USB Chager

// It works and It is stable. The issue is that I can't see serial traces un this way
```

Case B (firmware with Wifi connection)

```
// Case B.1
Swifitch <-> NodeMCU <-> Laptop USB

// It does NOT work, it reboots all the time

---

// Case B.2
Swifitch <-> NodeMCU <-> Mobile USB Chager

// It works and It is stable. The issue is that I can't see serial traces un this way
```

The only explanation that I've found os that the current supplied by my laptop USB is not enough.

ArnieX, 2018-03-25:
"The only explanation that I've found is that the current supplied by my laptop USB is not enough."

That could very well be the issue, my recommendation is to get USB doctor: https://www.aliexpress.com/item/Digital-Dispay-3V-30V-Mini-Current-Voltage-Charger-Capacity-Tester-USB-Doctor-qc2-0-3-0/32686482158.html?spm=2114.search0104.3.37.3cba7553yODxyV&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10151_10065_10068_10344_10342_10343_10340_10341_10696_10084_10083_10618_10304_10307_10302_5711215_10313_10059_10534_100031_10103_10624_10623_10622_10621_10620,searchweb201603_1,ppcSwitch_4&algo_expid=4fe0352c-214c-4ea8-aa4e-f8a4120dfb21-8&algo_pvid=4fe0352c-214c-4ea8-aa4e-f8a4120dfb21&priceBeautifyAB=0

dicastro, 2018-03-26:
This morning I've done a test with an iMac at work. It works better, I am able to connect to the swificth AP. But it is not stable enough, when I toggle the relay it restarts. So I think there is not current enough yet.

What I've tried is to have swificht powered to the main current and to the usb at same time.

```
Swifitch <-> NodeMCU <-> Laptop USB
    ^
    |
    +--> Main current (220 V AC)
```

**Can this option damage the swifitch?**

ArnieX, 2018-03-26:
OUCH!!! Have your read documentation properly? NEVER EVER connect mains and computer simultaneously. It may not hurt anything, but it is at least very bad idea. It could more likely destroy your computer than swifitch but still, please, don't do this. For your safety.

dicastro, 2018-03-26:
I did not remember that detail :-P

Nothing has exploded! But I am not going to do it again.

I am considering to buy this [dual usb cable](https://www.amazon.es/dp/B0047AALW6/?coliid=I1ZUP936VAUPJL&colid=1WFY4HLP30GFM&psc=0&ref_=lv_ov_lig_dp_it) to have extra power.

I've also bought the usb doctor that you recommended me to see the differences between the current of mac's usb and the current of windows/linux's usb

ArnieX, 2018-03-26:
That cable looks like really good idea ;)

dicastro, 2018-03-27:
With the _dual usb cable_, and connecting the serial wire to my laptop and the power wire to a mobile phone charger, it works like a charm!

> I have had some issue using a charger of an android device (2A charger of a samsung tablet), it wasn't stable. Finally I've used a 1A Iphone mobile charger. I do not understand why.

Thank you very much for your help!

closed 2018-03-27 by dicastro