ArnieX/swifitch #12: Can swifitch relay work with DC?

dicastro, 2018-03-21:
I do not know if this is an stupid question. I would like to know if it is possible to power swifitch with AC and use the relay for a DC circuit.

I have a really simple intercom at home and I am wondering if I could use swifitch to bypass the button of the intercom. Could I power swifitch, puggling it to main current (220V AC), and use swifitch's relay to bypass the button of the intercom, which works with DC? (it has 8.8 V and 65 mA)

ArnieX, 2018-03-21:
No because input voltage is also output voltage, it is connected on the board. So NO. But for DC transistor is better anyways. You can build simple circuit with the same transistor BCM817 that is used on swifitch, control it with swifitch GPIO and you will control this button. But I can't tell you how exactly to make this circuit because well I do not know myself :D.

dicastro, 2018-03-21:
Ok. Thank you!

I am wondering also if you, the team that has designed/developped swifitch, would be willing to make an adhoc pcb design (payed, of course)?

I think that swifitch is really well designed and I am interested in developping other modules, but I do not have the hardware knowledge.

ArnieX, 2018-03-21:
What kind of stuff are you thinking about?

dicastro, 2018-03-23:
Sorry for the delay. I am thinking in several things related to domotics.

1. I would like to have a swifitch with two relays, but conserving the reduced size. At home I have some double switches to control the lights of different zones in same room. I would like to have a swifitch inside this switches (this double switch is the same size as the single switch).
1. I would like to have a swifitch powered with DC and with a "relay" for DC (I think that for DC transistors are the right option, instead of relays). The use case is to have a swifitch to control my home's intercom, to open the external door. This intercom is powered with DC (8.8 V and 0.65mA). I would like to power the swifitch with the intercom's power. I know that maybe 65mA is not enough, so I've thought that it could be possible to have the swifitch powered with a rechargeable battery, and use the power of the intercom to recharge the swifitch's battery. I am planning to deploy this solution in different places, and all places won't have the same conditions, I mean same voltage or current. So I wonder if it would be possible to have something that worked with a range of input voltage (ex: from 5V to 24V).
1. I would also like to have something to control motorized blinds. I have not investigated anything about this subject. I have some motorized blinds but I do not know how do they work. A simple relay should be enough? How to control the sense of movement? How about security mechanisms? What if my cats are in the middle while the bilnds are closing?

ArnieX, 2018-03-23:
Hi @dicastro 

1, Yeah that would be nice, no promises thought, we are working on something else right now. We may revisit swifitch at later date.

2, You could do that, 5V should be possible to directly connect to the regulator (ie output of AC/DC converter - which wouldn't be there), or you could use some small DC/DC converter which will allow you to turn swifitch directly to DC operated. You will need to set this regulator to output 5V, then you will have switchable input voltage on relay. So it could work anywhere from 7V to 28V. But you need also sufficient current on output, at least of 500mA@5V so that the ESP has enough power.

3, Usually relay will do as signal to the blinds. But you need to discuss this with manufacturer of your blinds.

closed 2018-03-24 by dicastro