ArnieX/swifitch #13: zapojenie swifitchu namiesto jednotlačítkového vypínača

pete111, 2019-07-28:
Dobrý deň,
rád by som zapojil swifitch namiesto klasického jednotlačítkového vypínača, ktorý otvára a zatvára prívodný a odvodný kábel svetla. Je to možné? Lebo som si pozeral obrázok swifitchu a obávam sa, že swifitch potrebuje sólo dráhu napájania a následne spína/rozpína svetelný okruh.
Ak by sa to dalo zapojit úspešne aj do mojej kabeláže, bolo by to perfektné. ďakujem vopred za poradenie

ArnieX, 2019-07-28:
Nelze. Swifitch pro svou práci potřebuje napájení a pro plnohodnotné napájení je potřeba mít pod vypínačem také pracovní nulu (obvykle modrý drát). Bez pracovní nuly není uzavřený okruh a tak elektřina neproudí. Pokud tam ale pracovní nula je, tak tohle použití možné je. Vypínač se připojí na kontakty na swifitchi místo spojování fáze a k přepínání bude docházet cvaknutím relátkem.

pete111, 2019-07-28:
Vďaka za odpoveď. Chcel som odfotiť ten pôvodný vypínač, ale som momentálne niekde inde. Tak posielam fotku z iného vypínača, ktorý tu mám k dispozícii. Tu si robím nádej, že by to mohlo ísť :)
[image]