ArnieX/swifitch #14: Odpájanie od wifi

dzajro2, 2019-10-26:
Ahoj,
postavil som si 2ks swifitchu a oba kúsky mi robia, že po nejakom čase sa odpoja od wifi a naspäť sa pripoja až po odpojení napájania. Odpája sa mi to cca po hodine možno aj skôr. Signál wifi mám dobrý. Nevieš poradiť kde by mohol byť problém?

ArnieX, 2019-10-26:
Ahoj,
přiznám se že nevím. Může to teoreticky být i problém spojený s wifi routerem, když se to stejně děje s dvěma kusy swifitchů. Máš možnost zkusit je připojit na jinou wifi co běží na jiném wifi routeru?

dzajro2, 2019-12-23:
Ahoj,
tak bolo to použitými knižnicami. Použil som knižnice z githubu a funguje to.

ArnieX, 2019-12-24:
Ahoj, díky za zpětnou vazbu. Jsem rád že se ti to podařilo vyřešit. Krásné vánoce!